ethz-asl/mav_active_3d_planning #18: FATAL:No unreal game running to connect to

takayama110055, 2020-01-22:
Excuse me. 

When I tried to run a demo, I caught this fatal error.

ROS_MASTER_URI=http://localhost:11311

process[gazebo-1]: started with pid [13933]
process[firefly/spawn_firefly-2]: started with pid [13938]
process[firefly/mav_nonlinear_mpc-3]: started with pid [13939]
process[firefly/PID_attitude_controller-4]: started with pid [13940]
process[firefly/odometry_noise_model-5]: started with pid [13957]
process[firefly/tf_camera_to_link-6]: started with pid [13968]
process[unreal/unreal_ros_client-7]: started with pid [13969]
process[unreal/unreal_sensor_model-8]: started with pid [13992]
process[unreal/simulation_manager-9]: started with pid [13995]
process[unreal/eval_data_node-10]: started with pid [13998]
process[planner/voxblox_node-11]: started with pid [14001]
process[planner/planner_node-12]: started with pid [14002]
process[rviz_voxblox-13]: started with pid [14005]
SpawnModel script started
[INFO] [1579703247.561626, 0.000000]: Experiment setup: waiting for unreal MAV simulationto setup...
[ INFO] [1579703247.564671355]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1579703247.566875597]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[INFO] [1579703247.571044, 0.000000]: Loading model XML from ros parameter
[INFO] [1579703247.574111, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[INFO] [1579703247.587560, 0.000000]: Waiting for unreal camera params at '/unreal/unreal_ros_client/camera_params' ...
[INFO] [1579703247.634586, 0.000000]: Starting unreal MAV simulation setup coordination...
ERROR:__init__:205:Can not connect to ('localhost', 9000)
ERROR:__init__:206:Error [Errno 111] Connection refused
[FATAL] [1579703247.643283, 0.000000]: No unreal game running to connect to. Please start a game before launching the node.
ERROR:__init__:153:Fail to send message, client is not connected
[FATAL] [1579703247.643424, 0.000000]: Error addressing the unrealcv client. Try restarting the game.

I already installed UE4.

Could you please tell me what this may be due to? Thanks,

Schmluk, 2020-01-27:
In order to connect and exchange data with unreal engine, the unreal_cv Plugin for UE4 is used. Have you installed the Plugin to the Engine/Project you are using? More info about troubleshooting with UE4/unreal_cv can be found in the simulator package [unreal_cv_ros](https://github.com/ethz-asl/unreal_cv_ros).

closed 2020-01-31 by Schmluk

mnksnz, 2020-05-08:
Hi, 
I am facing the same issue and unable to install the unrealcv plugin. Could you please tell us which version of Unreal Engine and unrealcv plugin you are using? I am using Unreal Engine 4.25 and unrealcv 0.4.0 and somehow unrealcv plugin fails to builds. 

Thanks a lot!

Schmluk, 2020-05-08:
Unfortunately, we can only recommend to search the UnrealCV issues on why the plugin is not building.
The planning package does not directly use UnrealCV, but the [unreal_cv_ros](https://github.com/ethz-asl/unreal_cv_ros) simulator (which is based on UnrealCV with some additional commands). The simulator was developed and tested on UE4.16 with UnrealCV 0.3.10. For additional information on how to setup and modify the UnrealCV Plugin please see the docs in unreal_cv_ros. 
We tested UnrealCV up to UE4.18 but we can make no guarantees about the compatibility of the UnrealCV plugin with other UE4 versions.

mnksnz, 2020-05-13:
Thanks a lot for your reply. It helped a lot. I got it to work with versions 4.18 and 0.3.10. 4.16 somehow didn't work for me.