ethz-asl/mav_active_3d_planning #19: ImportError: No module named mav_active_3d_planning.msg

Hesitater, 2020-02-14:
Hello,

I got this error while executing ```roslaunch active_3d_planning_app_reconstruction example.launch``` following steps in your readme

```
...
Traceback (most recent call last):
  File "~/catkin_ws/src/mav_active_3d_planning/active_3d_planning_app_reconstruction/src/experiments/gps_odometry_simulator.py", line 11, in <module>
    from mav_active_3d_planning.msg import OdometryOffset
ImportError: No module named mav_active_3d_planning.msg
[firefly/odometry_noise_model-6] process has died 
...
```

Packages are successfully built on Ubuntu 18.04, ROS melodic. And I have checked the connection to unreal_cv_ros game: ```INFO:__init__:211:Got connection confirm: 'connected to RealisticRendering'```

The planner starts running afterwards but got this error

```
...
[INFO] [1581651386.915713, 19.518000]: Sensor model setup cleanly.
[INFO] [1581651386.941145, 19.544000]: unreal_ros_client is ready in fast mode.
[ERROR] [1581651387.397414, 20.000000]: Unrealcv error: error Can not find a handler for URI 'vget /uecvros/full -106.195015 437.424001 96.841962 -0.000001 -91.999411 -0.000073 1.000000 0'
[ERROR] [1581651387.616679, 20.220000]: Unrealcv error: error Can not find a handler for URI 'vget /uecvros/full -106.195015 437.424001 96.842092 0.000001 -91.999411 -0.000074 1.000000 0'
[ERROR] [1581651387.852067, 20.454000]: Unrealcv error: error Can not find a handler for URI 'vget /uecvros/full -106.195015 437.424001 96.842215 -0.000000 -91.999411 -0.000073 1.000000 0'
...
```
I'm not sure if these errors are related.

Schmluk, 2020-02-18:
Thank you for pointing out this error (which was due to renaming), it has been fixed as of #20 .

The second error is due to the fact that the RealisticRendering demo provided by UnrealCV does not contain the augmented Plugin. You can either run RealisticRendering in `standard` mode or run a world with the full Plugin (e.g. CityBuilding) in `fast` mode.

Hesitater, 2020-02-19:
Great! Now it works.

closed 2020-02-19 by Hesitater