ethz-asl/mav_active_3d_planning #21: Adding multiple sensors

biancabndris, 2020-05-12:
Hello, 
First of all, thank you for sharing this great code with the community, everything works perfectly. 
My question for you is which would be the way to go to extend the current planner to use multiple sensor modules? I really appreciate the modular structure of the code, and I understand that multiple instances of the sensor_module could perhaps be created, but I do not see how these could have different characteristics such as mounting position or resolution. 

Any help in this direction will be most welcomed. 

Thanks in advance!

Schmluk, 2020-05-12:
The sensors are used by the trajectory evaluators to identify visible voxels.

-  If you only need a specific evaluator and set of sensors, a fast way to add this would be to use multiple sensors directly in that evaluator.
- A more proper and modular way to do this would be to e.g. define a 'sensor_collection' sensor, that has the interface of a sensor but calls all its individual sensors to return the visible voxels to the evaluator.

I hope this helps, also feel free to open a PR if you decide to add this functionality.

biancabndris, 2020-05-18:
Thank you for the quick answer. I went for the fast way, but I am going to improve it and migrate towards a more proper solution as you suggested.

closed 2020-05-18 by biancabndris