ethz-asl/mav_active_3d_planning #22: Volumetric Indoor Exploration Scenario

Anzzy30, 2020-05-24:
Hello,

Thanks for your project it works perfectly.
I would like to know if you can provide the planner cfg used in your paper for the maze scenario?
Actually, I'm struggling a bit with all options available.

Schmluk, 2020-05-25:
Hi,

It is identical to the reconstruction_planner, just change 'VoxelWeightEvaluator' in line 69 to 'NaiveEvaluator' to use the number of unknown voxels as gain.

Also in the wiki, there is a short overview of the architecture and the modules available to construct a planer. When launched, all available params for the selected modules are printed to console, so you can see what these params and their values are.

Hope that helps!

Anzzy30, 2020-05-25:
Thank you for your answer!

Anzzy30, 2020-05-26:
Hi,

Sorry for reopening the issue but the planner node crash after some iteration (~1min running) with reconstruction_planner.yaml

The segfault occur here:
https://github.com/ethz-asl/mav_active_3d_planning/blob/c3708c8af5eaeb42c75f654adef93bdc90c54b2a/active_3d_planning_core/src/planner/online_planner.cpp#L264

> Thread 1 "reconstruction_" received signal SIGSEGV, Segmentation fault.
0x00007ffff7bc4323 in active_3d_planning::ros::RosPlanner::requestMovement(std::vector<active_3d_planning::EigenTrajectoryPoint, Eigen::aligned_allocator<active_3d_planning::EigenTrajectoryPoint> > const&) ()

Caused by trajectory, which is empty.
Do you have an idea how to resolve this error?

Schmluk, 2020-05-27:
Usually the trajectories should not be empty by construction. Could add a catch statement here that if the trajectory is empty we proceed with the backtracker or similar.

What TrajectoryGenerators are you using? Try removing the 'Feasible' from TrajectoryGenerator and BackTracker in the config if you use these, since the behavior is the same in simulation and the Feasible Versions use an external path generation library that can fail to produce a result I think.

Anzzy30, 2020-05-28:
I'm using reconstruction_planner.yaml, so RRTStar as TrajectoiryGenerator.
I tried to add an if statement, but the segfault appear in backtrack.

In rviz i observe weird behavior with the tree:

Here everything is fine, i guess.
[image]

And after more iteration i got this
[image]
where some node have no connection and invalid floating point error in PlannerVisualization.

I'm running it with a laptop: Ubuntu 16.04/ ROS Kinetic, i7 8750H (2.2GHz) and  Nvidia GTX 1070 MAXQ
Is performance or ros version could be the issue?

Schmluk, 2020-05-31:
Hi Anzzy,

Unfortunately, I don't seem to be able to reproduce your error. Performance should not be an issue since the planner simply produces as many segments as possible and it looks reasonable in your images. The framework was developed on a similar setup as yours, also on 16.04+kinetic.

The error appears to be indeed that some of the paths have zero points in them (which gives a div by zero error for the display text). Did you change anything else? Have you tried resetting to the current master branch? Otherwise could you provide the full config/stack trace so I can maybe help you further?

Anzzy30, 2020-06-01:
Hello,

Here is the console output before crash -> https://pastebin.com/vajXUPHs

I tried resetting to master branch nothing change when i use reconstruction_planner.yaml as cfg, crash happen.
This week i'll try a fresh install on an other computer if we cannot solve the issue.

ZbyLGsc, 2020-07-24:
@Anzzy30 @Schmluk I encountered a similar problem. I run the 'example.launch' with the Maze game. The only change I made is using RRTStar instead of Uniform TrajectoryGenerator. The planner crashes after about 1min.

Schmluk, 2020-07-29:
Dear @Anzzy30 @ZbyLGsc ,

We were finally able to track down the bug you were occasionally experiencing, it should be fixed as of #24. If it still crashes please feel free to re-open this issue.

Thank you for your patience and best wishes.

closed 2020-07-29 by Schmluk