ethz-asl/mav_active_3d_planning #23: Not able to build all packages

LuLiu21, 2020-06-07:
Hello everyone!
I am an amateur in ROS and trying to work with the mav_active_3d_planning.
I followed the instruction in the readme and when i run catkin build, i got this error. Any suggestions on how to solve this one?
I am using Ubuntu 16.04 and ros-kinetic.    
Thank you for any help :)

Errors     << mavros:make /home/liulu/catkin_ws/logs/mavros/build.make.000.log                  
/home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:332:72: error: ‘mavlink::omav’ has not been declared
  void handle_dynamixel(const mavlink::mavlink_message_t *msg, mavlink::omav::msg::DYNAMIXEL_STATUS &dyn_s)
                                                                        ^
/home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:332:100: error: expected ‘,’ or ‘...’ before ‘&’ token
  void handle_dynamixel(const mavlink::mavlink_message_t *msg, mavlink::omav::msg::DYNAMIXEL_STATUS &dyn_s)
                                                                                                    ^
/home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp: In member function ‘virtual mavros::plugin::PluginBase::Subscriptions mavros::std_plugins::IMUPlugin::get_subscriptions()’:
/home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:110:52: error: no matching function for call to ‘mavros::std_plugins::IMUPlugin::make_handler(void (mavros::std_plugins::IMUPlugin::*)(const mavlink_message_t*, int))’
           make_handler(&IMUPlugin::handle_dynamixel),
                                                    ^
In file included from /home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:19:0:
/home/liulu/catkin_ws/src/mavros/mavros/include/mavros/mavros_plugin.h:88:14: note: candidate: template<class _C> mavros::plugin::PluginBase::HandlerInfo mavros::plugin::PluginBase::make_handler(mavlink::msgid_t, void (_C::*)(const mavlink_message_t*, mavconn::Framing))
  HandlerInfo make_handler(const mavlink::msgid_t id, void (_C::*fn)(const mavlink::mavlink_message_t *msg, const mavconn::Framing framing)) {
              ^
/home/liulu/catkin_ws/src/mavros/mavros/include/mavros/mavros_plugin.h:88:14: note:   template argument deduction/substitution failed:
/home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:110:52: note:   cannot convert ‘&mavros::std_plugins::IMUPlugin::handle_dynamixel’ (type ‘void (mavros::std_plugins::IMUPlugin::*)(const mavlink_message_t*, int) {aka void (mavros::std_plugins::IMUPlugin::*)(const mavlink::__mavlink_message*, int)}’) to type ‘mavlink::msgid_t {aka unsigned int}’
           make_handler(&IMUPlugin::handle_dynamixel),
                                                    ^
In file included from /home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:19:0:
/home/liulu/catkin_ws/src/mavros/mavros/include/mavros/mavros_plugin.h:101:14: note: candidate: template<class _C, class _T> mavros::plugin::PluginBase::HandlerInfo mavros::plugin::PluginBase::make_handler(void (_C::*)(const mavlink_message_t*, _T&))
  HandlerInfo make_handler(void (_C::*fn)(const mavlink::mavlink_message_t*, _T&)) {
              ^
/home/liulu/catkin_ws/src/mavros/mavros/include/mavros/mavros_plugin.h:101:14: note:   template argument deduction/substitution failed:
/home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:110:52: note:   mismatched types ‘_T&’ and ‘int’
           make_handler(&IMUPlugin::handle_dynamixel),
                                                    ^
/home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:111:3: error: could not convert ‘{mavros::plugin::PluginBase::make_handler(void (_C::*)(const mavlink_message_t*, _T&)) [with _C = mavros::std_plugins::IMUPlugin; _T = mavlink::common::msg::ATTITUDE; mavros::plugin::PluginBase::HandlerInfo = std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> >; mavlink::mavlink_message_t = mavlink::__mavlink_message](&mavros::std_plugins::IMUPlugin::handle_attitude), mavros::plugin::PluginBase::make_handler(void (_C::*)(const mavlink_message_t*, _T&)) [with _C = mavros::std_plugins::IMUPlugin; _T = mavlink::common::msg::ATTITUDE_QUATERNION; mavros::plugin::PluginBase::HandlerInfo = std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> >; mavlink::mavlink_message_t = mavlink::__mavlink_message](&mavros::std_plugins::IMUPlugin::handle_attitude_quaternion), mavros::plugin::PluginBase::make_handler(void (_C::*)(const mavlink_message_t*, _T&)) [with _C = mavros::std_plugins::IMUPlugin; _T = mavlink::common::msg::HIGHRES_IMU; mavros::plugin::PluginBase::HandlerInfo = std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> >; mavlink::mavlink_message_t = mavlink::__mavlink_message](&mavros::std_plugins::IMUPlugin::handle_highres_imu), mavros::plugin::PluginBase::make_handler(void (_C::*)(const mavlink_message_t*, _T&)) [with _C = mavros::std_plugins::IMUPlugin; _T = mavlink::common::msg::RAW_IMU; mavros::plugin::PluginBase::HandlerInfo = std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> >; mavlink::mavlink_message_t = mavlink::__mavlink_message](&mavros::std_plugins::IMUPlugin::handle_raw_imu), mavros::plugin::PluginBase::make_handler(void (_C::*)(const mavlink_message_t*, _T&)) [with _C = mavros::std_plugins::IMUPlugin; _T = mavlink::common::msg::SCALED_IMU; mavros::plugin::PluginBase::HandlerInfo = std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> >; mavlink::mavlink_message_t = mavlink::__mavlink_message](&mavros::std_plugins::IMUPlugin::handle_scaled_imu), mavros::plugin::PluginBase::make_handler(void (_C::*)(const mavlink_message_t*, _T&)) [with _C = mavros::std_plugins::IMUPlugin; _T = mavlink::common::msg::SCALED_PRESSURE; mavros::plugin::PluginBase::HandlerInfo = std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> >; mavlink::mavlink_message_t = mavlink::__mavlink_message](&mavros::std_plugins::IMUPlugin::handle_scaled_pressure), <expression error>}’ from ‘<brace-enclosed initializer list>’ to ‘mavros::plugin::PluginBase::Subscriptions {aka std::vector<std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> > >}’
   };
   ^
/home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp: In member function ‘void mavros::std_plugins::IMUPlugin::handle_dynamixel(const mavlink_message_t*, int)’:
/home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:335:61: error: ‘dyn_s’ was not declared in this scope
   dynamixel_status->header.stamp = m_uas->synchronise_stamp(dyn_s.time_boot_us);
                                                             ^
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/imu.cpp.o] Error 1
make[2]: *** 正在等待未完成的任务....
make[1]: *** [CMakeFiles/mavros_plugins.dir/all] Error 2
make: *** [all] Error 2

Schmluk, 2020-06-07:
This looks like mavros can not find a mavlink installation. Could you try installing mavros according to [these instructions](https://github.com/ethz-asl/mavros/tree/master/mavros#installation)? Sometimes also clearing the build folder (~/catkin_ws/build/mavros) can help. 

Also mavros_extras is not required for the pipeline, so you can `touch ~/catkin_ws/src/mavros/mavros_extras/CATKIN_IGNORE` if this package causes issues.

LuLiu21, 2020-06-14:
@Schmluk Thanks for your respond, i have solve that problem by updating gzebo7 to gazebo9.
But still have a problem when i run catkin build:
Errors     << voxblox:cmake /home/liulu/catkin_ws/logs/voxblox/build.cmake.000.log
CMake Error at /home/liulu/catkin_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:38 (find_package):
  By not providing "Findprotobuf_catkin.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "protobuf_catkin", but CMake did not find one.

  Could not find a package configuration file provided by "protobuf_catkin"
  with any of the following names:

    protobuf_catkinConfig.cmake
    protobuf_catkin-config.cmake

  Add the installation prefix of "protobuf_catkin" to CMAKE_PREFIX_PATH or
  set "protobuf_catkin_DIR" to a directory containing one of the above files.
  If "protobuf_catkin" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:5 (catkin_simple)

Schmluk, 2020-06-15:
This package must have slipped through our installation testing, although we tested it on a clean system install. You can find it here: [https://github.com/ethz-asl/protobuf_catkin](https://github.com/ethz-asl/protobuf_catkin). It has been added to the rosinstalls.

Thank you for your patience.

LuLiu21, 2020-06-23:
@Schmluk Thank you for your help!  The problem is solved.

But now I'm struggling with the data repository can not be downloaded successfully  from 
https://www.polybox.ethz.ch/index.php/s/6vhPDINcISbEogg 
I hope to get your further help!

LuLiu21, 2020-06-24:
I solved the download probolem by using ssr and wget. HAHA

Schmluk, 2020-06-24:
Good to hear, let us know if you run into further trouble.

closed 2020-06-24 by Schmluk

sushiye, 2021-03-05:
> Hello everyone!
> I am an amateur in ROS and trying to work with the mav_active_3d_planning.
> I followed the instruction in the readme and when i run catkin build, i got this error. Any suggestions on how to solve this one?
> I am using Ubuntu 16.04 and ros-kinetic.
> Thank you for any help :)
> 
> Errors << mavros:make /home/liulu/catkin_ws/logs/mavros/build.make.000.log
> /home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:332:72: error: ‘mavlink::omav’ has not been declared
> void handle_dynamixel(const mavlink::mavlink_message_t _msg, mavlink::omav::msg::DYNAMIXEL_STATUS &dyn_s) ^ /home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:332100 error: expected ‘,’ or ‘...’ before ‘&’ token void handle_dynamixel(const mavlink::mavlink_message_t _msg, mavlink::omav::msg::DYNAMIXEL_STATUS &dyn_s) ^ /home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp: In member function ‘virtual mavros::plugin::PluginBase::Subscriptions mavros::std_plugins::IMUPlugin::get_subscriptions()’: /home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:110:52: error: no matching function for call to ‘mavros::std_plugins::IMUPlugin::make_handler(void (mavros::std_plugins::IMUPlugin::_)(const mavlink_message_t_, int))’
> make_handler(&IMUPlugin::handle_dynamixel),
> ^
> In file included from /home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:19:0:
> /home/liulu/catkin_ws/src/mavros/mavros/include/mavros/mavros_plugin.h:88:14: note: candidate: template mavros::plugin::PluginBase::HandlerInfo mavros::plugin::PluginBase::make_handler(mavlink::msgid_t, void (_C::_)(const mavlink_message_t_, mavconn::Framing))
> HandlerInfo make_handler(const mavlink::msgid_t id, void (_C::_fn)(const mavlink::mavlink_message_t _msg, const mavconn::Framing framing)) { ^ /home/liulu/catkin_ws/src/mavros/mavros/include/mavros/mavros_plugin.h:88:14: note: template argument deduction/substitution failed: /home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:110:52: note: cannot convert ‘&mavros::std_plugins::IMUPlugin::handle_dynamixel’ (type ‘void (mavros::std_plugins::IMUPlugin::_)(const mavlink_message_t_, int) {aka void (mavros::std_plugins::IMUPlugin::_)(const mavlink::__mavlink_message_, int)}’) to type ‘mavlink::msgid_t {aka unsigned int}’
> make_handler(&IMUPlugin::handle_dynamixel),
> ^
> In file included from /home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:19:0:
> /home/liulu/catkin_ws/src/mavros/mavros/include/mavros/mavros_plugin.h:101:14: note: candidate: template<class _C, class _T> mavros::plugin::PluginBase::HandlerInfo mavros::plugin::PluginBase::make_handler(void (_C::_)(const mavlink_message_t_, _T&))
> HandlerInfo make_handler(void (_C::_fn)(const mavlink::mavlink_message_t_, _T&)) {
> ^
> /home/liulu/catkin_ws/src/mavros/mavros/include/mavros/mavros_plugin.h:101:14: note: template argument deduction/substitution failed:
> /home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:110:52: note: mismatched types ‘_T&’ and ‘int’
> make_handler(&IMUPlugin::handle_dynamixel),
> ^
> /home/liulu/catkin_ws/src/mavros/mavros/src/plugins/imu.cpp:111:3: error: could not convert ‘{mavros::plugin::PluginBase::make_handler(void (_C::_)(const mavlink_message_t_, _T&)) with _C = mavros::std_plugins::IMUPlugin; _T = mavlink::common::msg::ATTITUDE; mavros::plugin::PluginBase::HandlerInfo = std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> >; mavlink::mavlink_message_t = mavlink::__mavlink_message, mavros::plugin::PluginBase::make_handler(void (_C::_)(const mavlink_message_t_, _T&)) with _C = mavros::std_plugins::IMUPlugin; _T = mavlink::common::msg::ATTITUDE_QUATERNION; mavros::plugin::PluginBase::HandlerInfo = std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> >; mavlink::mavlink_message_t = mavlink::__mavlink_message, mavros::plugin::PluginBase::make_handler(void (_C::_)(const mavlink_message_t_, _T&)) with _C = mavros::std_plugins::IMUPlugin; _T = mavlink::common::msg::HIGHRES_IMU; mavros::plugin::PluginBase::HandlerInfo = std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> >; mavlink::mavlink_message_t = mavlink::__mavlink_message, mavros::plugin::PluginBase::make_handler(void (_C::_)(const mavlink_message_t_, _T&)) with _C = mavros::std_plugins::IMUPlugin; _T = mavlink::common::msg::RAW_IMU; mavros::plugin::PluginBase::HandlerInfo = std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> >; mavlink::mavlink_message_t = mavlink::__mavlink_message, mavros::plugin::PluginBase::make_handler(void (_C::_)(const mavlink_message_t_, _T&)) with _C = mavros::std_plugins::IMUPlugin; _T = mavlink::common::msg::SCALED_IMU; mavros::plugin::PluginBase::HandlerInfo = std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> >; mavlink::mavlink_message_t = mavlink::__mavlink_message, mavros::plugin::PluginBase::make_handler(void (_C::_)(const mavlink_message_t_, _T&)) with _C = mavros::std_plugins::IMUPlugin; _T = mavlink::common::msg::SCALED_PRESSURE; mavros::plugin::PluginBase::HandlerInfo = std::tuple<unsigned int, const char*, long unsigned int, std::function<void(const mavlink::__mavlink_message*, mavconn::Framing)> >; mavlink [... cut ...]
> dynamixel_status->header.stamp = m_uas->synchronise_stamp(dyn_s.time_boot_us);
> ^
> make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/imu.cpp.o] Error 1
> make[2]: *** 正在等待未完成的任务....
> make[1]: *** [CMakeFiles/mavros_plugins.dir/all] Error 2
> make: *** [all] Error 2

hello, i have the same problem like "mavlink::omav’ has not been declared" when i build the modified mavros of ethz-asl, could you please tell me the  detailed solution to solve it ? what's your mavlink link or version? thanks a lot!

Schmluk, 2021-03-05:
Hi @sushiye 

mavros (the package that fails) is a third party dependency, so I can not  tell you what is going wrong exactly. However, not all components of the dependencies are required for active_3d_planning. Can you try building only the required packages, i.e. `catkin build active_3d_planning_app_reconstruction`?

sushiye, 2021-03-08:
> Hi @sushiye
>
> mavros (the package that fails) is a third party dependency, so I can not tell you what is going wrong exactly. However, not all components of the dependencies are required for [...]

sushiye, 2021-03-08:
Hi @Schmluk 
thanks for your reply! i had the problem in another project and  i have solved the catkin build problem by deleting some parts of the modified mavros.