ethz-asl/mav_active_3d_planning #25: Error:  planner/voxblox_node-11

itsalpha, 2020-08-01:
Hello everyone!
I am using Ubuntu 16.04 and ros-kinetic, and sucessful build with readme.But I meet a Error when I launch the active_3d_planning_app_reconstruction example.launch.
'''
[planner/voxblox_node-11] process has died [pid 614, exit code 1, cmd /home/server2/catkin_mav/devel/lib/voxblox_ros/esdf_server -alsologtostderr pointcloud:=/unreal/unreal_sensor_model/ue_sensor_out __name:=voxblox_node __log:=/home/server2/.ros/log/408e070e-d393-11ea-a7eb-049226c0ea4e/planner-voxblox_node-11.log].
'''
Thanks for your help.

pamzerbhu, 2020-08-01:
The same mistake.

itsalpha, 2020-08-01:
And i want to know， 
What version of protobuf are you using?

closed 2020-08-01 by itsalpha

wangjinhoon, 2024-03-11:
how did you solve it

i met this error
```
ERROR: unknown command line flag 'alsologtostderr'

[planner/voxblox_node-11] process has died [pid 19485, exit code 1, cmd /home/jh/catkin_ws/devel/lib/voxblox_ros/esdf_server -alsologtostderr pointcloud:=/unreal/unreal_sensor_model/ue_sensor_out __name:=voxblox_node __log:=/home/jh/.ros/log/c294fffc-df72-11ee-ae12-00d861164570/planner-voxblox_node-11.log].
log file: /home/jh/.ros/log/c294fffc-df72-11ee-ae12-00d861164570/planner-voxblox_node-11*.log
```