ethz-asl/mav_active_3d_planning #26: A strange result output by run the experiment1.

pamzerbhu, 2020-08-08:
Hello, thanks for your project. It works very nice. 
     I run the run_experiment.launch directly but got a strange result after 30 min. Then, I run the evaluate_experiment.launch and got the result named ***error.ply. It performance seems not as good as in the paper.
[image]
Thanks for your help！

Schmluk, 2020-08-10:
Hi @pamzerbhu 

The following points may hopefully help you:
* From your image, it looks like the coordinate axes are not aligned: The coordinate frame of the simulation is initialized to zero when the simulation starts. For evaluation, we assume that the the simulation frame is identical with the frame, in which the game starts. If you happen to move the camera between starting the UE4 game and starting the simulation, the frames won't be aligned and the evaluation will look odd. Therefore you have to be careful to tab out of game control before moving. This is easiest in the UE4 Editor with Alt+P->Ctrl+Shift+F1.
* Please note that the planner in the demo is not the one used in the paper. The planner from the paper is given in `reconstruction_planner.yaml`.
* Last, the planner performance also depends on the hardware available, as it will simply expands as many segments as possible with 1 virtual CPU core. Therefore the results may vary between machines.

Please let me know if you have further questions or run into trouble.

pamzerbhu, 2020-08-11:
Hi @Schmluk 
Thanks for your reply, I carefully followed your advice and found this problem:

- step 1, Use OBS to record UE4 game and terminal image.

- step 2, I launch the UE game expriment1 and carefully switch to the terminal.

- step 3, Launch run_expriment.launch in terminal.
 In this step, until ROS information "*******planner is now running*********" come out, the simulation of UAV looks like coordinate axes are correct like this screenshot:
[image]
But the virtual UAV has changed in yaw when the rivz launch. Like this screenshot
[image]

I guess this is the reason that coordinate axes are not aligned. Holp could give me some help, Thanks again sincerely.

pamzerbhu, 2020-08-13:
I found a method that could solve this problem:

- run "roscore" command before launch "run_expriment.launch" and launch UE4 game.
- run UE4 game binary and then quickly launch the "run_expriment.launch" before game launch. 

By those two methods launch the demo. Axes will be correct in the final error map.
[image]

Schmluk, 2020-08-13:
Hi @pamzerbhu 

Great that you were able to resolve the issue. Although it seems pretty strange that this is necessary. If we find a way to fix this more reliably we will let you know.

Thank you and best wishes

closed 2020-09-23 by pamzerbhu