ethz-asl/mav_active_3d_planning #27: Keeps replanning in a custom simulation

LXYYY, 2020-11-18:
Hi @Schmluk , I am trying to adopt mav_active_3d_planning in our own work recently. I am running it in a gazebo world, instead of the unrealcv game. I changed the related input/output, and trying to make the planning work. But it stuck at the beginning printing this:
[image]
And the input tsdf map looks correct according to rviz:
[image]
Also the odometry topic should be correct since I echoed it and the rotors hovering example works correct.
[image]
experiment setting is 
```map_bounding_volume:
  x_min: -20.0
  x_max: 20.0
  y_min: -20.0
  y_max: 20.0
  z_min: 0
  z_max: 3

evaluation_bounding_volum:
  x_min: -20.0
  x_max: 20.0
  y_min: -20.0
  y_max: 20.0
  z_min: 0
  z_max: 3

target_bounding_volume:
  x_min: 0
  x_max: 15
  y_min: 0
  y_max: 15
  z_min: 1
  z_max: 2

# System Constraints
planner:
  system_constraints:
    v_max: 0.5
    a_max: 0.5
    yaw_rate_max: 1.57
```

Do you have any advice where I may do wrong?  wrong params or input maybe?

Schmluk, 2020-11-18:
Hi @LXYYY, 
In general the framework should be compatible with various simulators, we also ran it on different ones. What you are describing happens when the planner does not find any feasible paths, and since there are no prior paths to reverse it will essentially rotate for ever. There's different possible reasons for this:

* **The map does not contain enough data:** The map you are showing does not look like you're using the default voxblox map, or have you added and interface to octomap/fiesta? If you use voxblox for planning (the default code) could you visualize the 'traversaility' or 'tsdf' topic? For that to get published you need to set the params 'publish_traversable' or 'publish_pointclouds', respectively, to true in the voxblox node.

* **There is too much unobesrved space at the starting pose:** Since the robot usually starts in unobserved space there is a param 'clearing_radius' in the active_3d_planner config which treats unobserved space within that distance of the robot as traversable to help getting started, you can try setting it to a higher value.

* **The robot is outside the traversable bounds:** Everything outside the map_bounding_volume is intraversable. Notice that the planner typically operates in odom frame, i.e. the drone starts at the origin. In that case a bounding volume that starts at 0 is likely to not contain the drone which makes paths infeasible.

I hope this helps, let us know if you have more questions.

Best wishes,
Lukas

LXYYY, 2020-11-27:
Thanks for your advice @Schmluk , made it run after setting a bigger clearing radius

Schmluk, 2020-11-27:
Good to hear! Let us know if you run into more trouble.

closed 2020-11-27 by Schmluk

LXYYY, 2020-12-07:
Hi @Schmluk , after successfully tested mav_active_3dplanning with our work, I also want to use the evaluation app. But I am a liitle confused about how to connect that evaluation_data_node to gazebo world topics. And is this the case, if I do not run that evaluation node, the planner node itself won't generate enough evaluation data?

Badrivishal, 2023-04-10:
Hii, how did you get it to run in a gazebo environment?, I am unable to run it with gazebo. 
The planner sends this message after toggle, 
[ INFO] [1681126198.013734540, 1703.728000000]: Started planning.
[ INFO] [1681126198.210977368, 1703.920000000]: Replanning!
(0.216s elapsed, 3 new, 3 total, 1 killed by root change, 0 killed while updating)

After this message, nothing comes up. Nothing is happening..