ethz-asl/mav_active_3d_planning #28: How to use evaluation on a gazebo simulation

LXYYY, 2020-12-07:
Hi @Schmluk , after successfully tested mav_active_3dplanning with our work, I also want to use the evaluation app. But I am a liitle confused about how to connect that evaluation_data_node to gazebo world topics. And is this the case, if I do not run that evaluation node, the planner node itself won't generate enough evaluation data?

LXYYY, 2020-12-07:
And I also ran into this error when I tried to set a bigger `min_path_length`, because I am running it in a outdoor simulation, and 0.5 looks too small compared to environment size and mav speed. 
[image]
This error showed up both with min path length 5 and 10. How should I set params to make each path branch longer?

Schmluk, 2020-12-07:
Hi @LXYYY,

Good to hear. Regarding the first question, the planner node itself only stores data for the performance analysis (e.g. where how much time was spent). The eval_data node manages simulation setup and shutdown and periodically stores the map for later evaluation against ground truth pointclouds. To run it with Gazebo, the following topics need to be remapped:
* ue_out_in: Counts how many pointclouds were received. You can also leave this blank if it does not interest you.
* collision: When this topic receives a message (of type std_msgs/String) the experiment is shutdown and collision is entered to the data log so you can check what happened in your experiments. I don't know whether this information is readily available in your simulation, if the simulation cannot collide or detect collision you can also leave this empty and it will always run till maximum time is reached.

Regarding your second question, I think `max_path_length` is the param you want to increase, in any case make sure it is larger than min_path_length. If the error persists can you please add the full configuration to the issue?

Thank you and best wishes,
Lukas

LXYYY, 2020-12-08:
Thanks @Schmluk , a bigger `max_path_length` works for me. I will try the evaluation later. But I run into another error today, when I was trying to set trajectory generator type to `FeasibleRRTStar` to get smoother derivatives, it crashed:
[image]
and I print `trajectory.time_from_start_ns` before crash, it showed
[image]
and I tried to start planning when last trajectory generated by another node, both finished or unfinished, the error still exists. Any idea?

closed 2021-05-25 by Schmluk