ethz-asl/mav_active_3d_planning #31: Planner exploring areas other than the Building (CityBuilding Reconstruction Experiment)

manthan99, 2021-02-16:
Hii

First of all, thank you for the excellent software. I am trying to reconstruct the city building using the reconstruction planner using the default planner configurations. I observed that in some of the runs, if the planner starts to explore the flat land instead of the building, it keeps on exploring that portion instead of the building and fails to return back to the building. I am attaching a screenshot of the sample run-

[image]

This screenshot was taken after almost 11 minutes of runtime.

[image]

This screenshot was taken after almost 18 minutes of runtime.

Could you please suggest changing which parameters would be beneficial to avoid this from happening?

Secondly, what might be the easiest way to bound (limit) the exploration region so that we can reconstruct only the specific areas of interest instead of the entire map?

Schmluk, 2021-02-16:
Hi @manthan99 

Thank you for your interest. It is possible to set a region of interest (bounding box) for the planner via params, the configuration you are likely using is found here `active_3d_planning_app_reconstruction/cfg/experiments/CityBuilding.yaml`. An important thing to note is that the region of interest is always in Odometry frame of the robot, i.e. the origin is at the pose of the robot when the planner is launched with x pointing forward, y left, z up. This is likely also the reason why the robot explores the wrong area, since the current bounding box assumes the robot is at the initial pose of the simulation. If you move the robot (or the simulation via user control) in before the planner is started, these two frames won't align and the region of interest will be wrongly placed.

To fix this I recommend starting the simulation (without moving the mouse to guarantee you don't move the camera) and then tab out of game control (Alt+Tab for Binary, Ctrl+Shift+F1 for Editor). If you then start the planner the frames should match and the ROI is an enclosure of the building.

Hope this helps, let us know if you have more questions.

Schmluk, 2021-02-16:
Hi @manthan99 

Thank you for your interest. The planner uses a region of interest (bounding box) that can be set via params. The configuration you are likely using is found here `active_3d_planning_app_reconstruction/cfg/experiments/CityBuilding.yaml`. It is important to note that the ROI is given in Odometry frame, i.e. the origin is at the pose of the robot when the planner is launched with x point forward, y left and z up. The values given in the configuration expect the robot to start at the initial pose of the simulation, in which case the ROI encloses the building. 

If you move the robot (or the simulation via user control), these frames will not align and the ROI will be misplaced, which is likely why the planner focuses on the open space in your case. To fix this I recommend starting the UE4 simulation (without touching the mouse to make sure you don't interact with the simulation) and then tab out of game control (Alt+Tab for Binary, Ctrl+Shift+F1 for Editor). Afterwards you can launch the planner and it should focus on the building.

Hope this helps, let us know if you have more questions.

manthan99, 2021-02-16:
Thankyou, this solves the problem.

closed 2021-02-16 by manthan99