ethz-asl/mav_active_3d_planning #32: Evaluation of Custom models

manthan99, 2021-02-17:
Hii

Once again thank you for the amazing package. I am trying to run experiments on my custom scenario which I created using the instruction given [here](https://github.com/ethz-asl/unreal_cv_ros#UnrealCV-Plugin-Setup). Now I was successfully able to run the experiment by specifying the bounds in a separate YAML file. You can see the reproduced mesh below-
[image]
[image]
I am facing problems during the evaluation part. As you can see in the picture above, the majority of the surfaces remain grey are not able to be matched. Most probably, this is due to the orientation error which might have occurred while producing the groundtruth point cloud. I ensured that the UE4 game is instantly minimized on startup and the camera angle is not moved. I did the following steps to produce the groundtruth point cloud-

- In the Unreal Engine Editor,

1. Select the objects of interest in the World Outliner
2. File > Export Selected...
3. Save as *.obj file. (Currently no need to export the material too.)

- Open CloudCompare,

4. File > Import > my_mesh_export.obj
5. Use 'Edit > Multiply/Scale > 0.01' to compensate for unreal engine units (default is cm).
6. Use 'Edit > Apply transformation' to place and rotate the object relative to the PlayerStart settings from unreal. (I.e. with the origin at the PlayerStart location, x pointing towards the PlayerStart orientation, y-left and z-up). I first rotated the object 90 degrees w.r.t. X-axis and then -90 degrees w.r.t Z axis. The resulting orientation seemed to be correct.
7. Click 'Sample points on a mesh' to create a pointcloud. (I sampled 5000 points density)
8. 'File > Save' and save as a *.ply in ASCII format

- With a TextEditor,

9. Open my_gt_pointcloud.ply and remove the "comment Author" and "obj_info" fields (lines 3 and 4, these create errors with pcl::plyreader).

I am unable to figure out what might be the issue. I tried the same thing with the CityBuilding model as well. If I use the default groundtruth ply file provided, the results are perfect, but if I follow the above method, I am getting similar incorrect results. [Images attached below]
[image]
[image]

manthan99, 2021-02-17:
I was able to solve this more or less by applying transformations (translation) in x, y, and z as per the player's initial start position. Though there were offsets in all 3 directions which had to be set manually by observation. Could you suggest the exact method for finding this x,y and z (global box center as per Cloud compare) ?

Schmluk, 2021-02-18:
Hi @manthan99 

If the ground truth pointcloud pose does not match the one expected for the scan you can e.g. align it to a mesh obtained using ground truth scans (no corruptions). Either manually or using tools such as [align](https://www.cloudcompare.org/doc/wiki/index.php?title=Align) or [ICP](https://www.cloudcompare.org/doc/wiki/index.php?title=ICP).

manthan99, 2021-02-18:
Great Thankyou very much! 
This is exactly what I was asking for.

closed 2021-02-18 by manthan99