ethz-asl/mav_active_3d_planning #36: Recreating Maze Experiment from Paper

danielduberg, 2021-03-12:
Hi,
First of all, thank you for this nice project and paper!

The last week I have been trying to recreate your Maze experiment from your paper. However, I never manage to make the MAV explore the whole environment and after a while, it deletes the whole tree. I read in another issue (#22) that you used the `reconstruction_planner.yaml` file for the Maze experiment as well, the only difference being that you changed `VoxelWeightEvaluator` to `NaiveEvaluator`, so I did that.

Below you can see the config file I used
[test.txt](https://github.com/ethz-asl/mav_active_3d_planning/files/6129804/test.txt)

I also recorded a video so you can see what is happening. The video is speed-up 2X.
https://youtu.be/n31gT60CPY4

Any help is very appreciated, thank you!

Schmluk, 2021-03-15:
Hi @danielduberg 

Thank you for your interest. Judging from your video it appears that the gain computation does not work as intended (since the majority of them seem to be 0). Could  you enable all visualizations of the gain (including the text + views), and see which voxels are counted? 

One common source of errors here is that the target bounding volume is expected in odometry frame, i.e. with respect to the drone's start position. If the drone does not start at [0,0,0], the bounding box needs to be adjusted for it.

Regarding parts of the tree dying, this can happen when the map changes in such a way that segments are no longer traversable. In particular if the robot moves close to a wall it can be that it's connection to the tree becomes infeasible, essentially cutting off the whole tree. However, this should usually not be a problem since it's quickly rebuilt.

Best wishes,
Lukas

danielduberg, 2021-03-15:
Thank you @Schmluk, I will look into the target bounding volume.

I was also thinking it has something to do with the gain computation. I will enable all visualizations of the gain and record a new video. I have looked at it before but I was not sure what it should look like.

Thank you very much for your help!

danielduberg, 2021-03-15:
Here is a video where all visualizations of the gain (including the text + views) are enabled.
https://youtu.be/QFRZvOeyqBY

Here is the same video at 2x speed 
https://youtu.be/T6LpHpWSj60

I use the same configuration as before. I started the exploration process at [0, 0, 0], meaning the drone spawned at [0, 0, 0] and I called the `/planner/planner_node/toggle_running` service when the drone was at [0, 0, 0].

Schmluk, 2021-03-16:
There still seems to be something going wrong with the gain computations. E.g. at the beginning (and at other times) many gains are zero which they definitively shouldn't be but at some points it looks like it's doing what it's supposed to. 

To make sure the bounding volume is not the culprit here you can try to set everything to +/- a large value except for z-max?

As a second verifiation, are you using voxblox as map representation? If so you could try to visualize the voxblox server that is inside the planner (it only subscribes to map updates from the /planner/voxblox_node and uses it for lookups) to check that the map available to the planner makes sense.

danielduberg, 2021-03-16:
Yes, I am using Voxblox as map representation. I will try to visualize the voxblox server inside the planner. I will also increase the size of the bounding volume as you proposed.

Thank you, I will come back when I have new results.

danielduberg, 2021-03-16:
Unfortunately, increasing the bounding volume does not seem to help.

Here is a video visualizing the different `planner_node` topics at the beginning.
https://youtu.be/TV821S9zktY
Does this look okay to you? 

I used a trajectory_generator, clearing_radius of 1.0 in this case.

Schmluk, 2021-03-16:
Hi @danielduberg,

I double checked some potential issues, could you checkout #37 and check whether it solves your problem?
It worked fine on my machine. I also added an example config for the exploration planner (which should be extremely similar to yours).

Some other considerations I noticed regarding performance / strange behavior in your setup:
* The planner uses the ESDF map for lookups, if you use the 'clear_spehere_for_planning' in voxblox, the esdf around the starting position will be filled in with large values. This explains why there were no gains where the planner started, even though there is no mesh there. I would recommend not using this voxblox param but set the 'clearing_radius' in the planner config to e.g. 1.0.
* There is a bit of latency between sensor data being integrated, published to the second voxblox server and the gains being updated in the planner. Since you move very quickly between viewpoints this latency might induce strange behavior.
* Similarly, the number of new samples is dependent on the used hardware and time available. If the drone moves very fast it can occur that too few local samples are added to continue exploration. If these two points are an issue for your case you can try setting the replan_pos/yaw_threshold of the planner lower or enforce a minimum number of samples by setting min_new_tries/segments.

Let me know if this helps.

danielduberg, 2021-03-16:
Big thanks @Schmluk!

I will look into these things and try the things you proposed first thing tomorrow morning.

Also, thank you for the detailed explanation. My previous runs have been on an Intel Core i7 8700K with 64 GB of RAM and an AMD Ryzen 5 3600x with 32 GB of RAM, so I think hardware wise it should be fine.

closed 2021-05-25 by Schmluk