ethz-asl/mav_active_3d_planning #47: libmav_nonlinear_mpc_lib.so: undefined reference to `LAPACKE_dgees

ArghyaChatterjee, 2022-01-14:
Hi, I was trying to build the workspace in my ubuntu 18.04 (ROS Melodic) while I encountered this error. Here is the terminal output.
```
Errors     << mav_nonlinear_mpc:make /home/arghya/mav_active_3d_ws/logs/mav_nonlinear_mpc/build.make.002.log                                                           
/home/arghya/mav_active_3d_ws/devel/lib/libmav_nonlinear_mpc_lib.so: undefined reference to `LAPACKE_dgees'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/arghya/mav_active_3d_ws/devel/lib/mav_nonlinear_mpc/nonlinear_mpc_node] Error 1
make[1]: *** [CMakeFiles/nonlinear_mpc_node.dir/all] Error 2
make: *** [all] Error 2
```
Is there any solution to this problem ??

Schmluk, 2022-01-14:
Hi @ArghyaChatterjee 

This looks like a dependency of mav_nonlinear_mpc is missing, `sudo apt-get install liblapacke-dev` should fix this. Otherwise, you can check the install instructions or issues on https://github.com/ethz-asl/mav_control_rw, these might have more relevant insights.

Schmluk, 2022-02-22:
Closed due to inactivity.

closed 2022-02-22 by Schmluk

cwy-16, 2022-07-26:
> 嗨，当我遇到此错误时，我试图在我的ubuntu 18.04（ROS Melodic）中构建工作区。这是终端输出。
> 
> ```
> Errors     << mav_nonlinear_mpc:make /home/arghya/mav_active_3d_ws/logs/mav_nonlinear_mpc/build.make.002.log                                                           
> /home/arghya/mav_active_3d_ws/devel/lib/libmav_nonlinear_mpc_lib.so: undefined reference to `LAPACKE_dgees'
> collect2: error: ld returned 1 exit status
> make[2]: *** [/home/arghya/mav_active_3d_ws/devel/lib/mav_nonlinear_mpc/nonlinear_mpc_node] Error 1
> make[1]: *** [CMakeFiles/nonlinear_mpc_node.dir/all] Error 2
> make: *** [all] Error 2
> ```
> 
> 这个问题有什么解决方案吗？

请问您这个问题是怎么解决的呢？