ethz-asl/mav_active_3d_planning #48: How to launch the "FeasibleRRTstar" ?

9woods123, 2022-03-12:
Hello！Thank you for opening such a great job！ I am trying to modify the planner_config （.yaml file）for using "FeasibleRRTstar" which is mentioned in the wiki. But when I  modify the type from "RRTstar" to the "FeasibleRRTstar", the program cannot proceed.
             If you have a planner_config for "FeasibleRRTstar", could you please upload it?  Looking for your kind reply.

Schmluk, 2022-03-14:
Hi @9woods123 

Thanks a lot for your interest! The main issue with FeasibleRRTStar is a problem with a downstream package (https://github.com/ethz-asl/mav_trajectory_generation) as far as I know, which we currently don't have the resources to fix.

If you use the regular RRTStar the outcome should be almost exactly the same in simulation and on a real robot with a decent controller (we tested this on one of our hexacopters too). So for the moment I'd recommend just sticking to RRTStar.

Hope this helps, let us know if you have more questions.

Schmluk, 2022-03-14:
Great to hear! Please feel free to open a PR for this fix, this might also be useful for others!

Schmluk, 2022-07-05:
Thank you for your investigations and contribution! This issue should be fixed as of #50 .

closed 2022-07-05 by Schmluk