ethz-asl/mav_active_3d_planning #52: Compilation failed

MaLin-art, 2022-07-24:
My computer is Ubuntu18.04+ROS Melodic,According to the readme, at the "catkin build" step, it fails every time：
   [build] Summary: 25 of 52 packages succeeded
   [build] Ignored: 3 packages were skipped or are blacklisted
   [build] Warning: 6 packages succeeded with warnings 
   [build] Abandoned: 24 packages were abandoned
   [build] Failed: 3 packages failed.

Schmluk, 2022-07-24:
Hi @MaLin-art 
Which packages fail to build? Can you paste the full compilation output?

MaLin-art, 2022-07-25:
I am very happy to receive your email. I am a student from Hong Kong. I really appreciate your help. This is my compilation process. Thank you for your guidance!

Schmluk, 2023-01-12:
Closed due to inactivity/missing info.

closed 2023-01-12 by Schmluk