ethz-asl/mav_active_3d_planning #62: There are four errors in the first running code of the configuration planner in the example.

zcc0514, 2024-01-12:
Hello!
First of all, thank you very much for your wonderful work! I have completed the installation process described in your README.md, and I can also successfully download experimental scenes, ground truth point clouds and other related resources. However, in the following example-configuring Planner, there was a problem when running "roslaunch active_3d_planning_app_reconstruction example.launch", and four errors were generated. The following is the result after I entered this line of command, and I will send you a separate screenshot of the error report. I would be very grateful if you could answer it for me. Looking forward to your reply!

... logging to /home/dlvirmac/.ros/log/d4b84872-b13e-11ee-9e6e-525400d9ad51/roslaunch-dlvirmac-Standard-PC-i440FX-PIIX-1996-27030.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dlvirmac-Standard-PC-i440FX-PIIX-1996:36105/

SUMMARY
========

CLEAR PARAMETERS
 * /firefly/PID_attitude_controller/
 * /firefly/mav_nonlinear_mpc/

PARAMETERS
 * /firefly/PID_attitude_controller/allocation_matrix: [0.5, -0.866025, ...
 * /firefly/PID_attitude_controller/arm_length: 0.2156
 * /firefly/PID_attitude_controller/inertia: [0.034756, 0.0, 0...
 * /firefly/PID_attitude_controller/max_integrator_error: 0.6
 * /firefly/PID_attitude_controller/n_rotors: 6
 * /firefly/PID_attitude_controller/p_gain: 25
 * /firefly/PID_attitude_controller/pitch_gain: 70
 * /firefly/PID_attitude_controller/pitch_int_gain: 2.0
 * /firefly/PID_attitude_controller/q_gain: 22
 * /firefly/PID_attitude_controller/r_gain: 1.0
 * /firefly/PID_attitude_controller/roll_gain: 70
 * /firefly/PID_attitude_controller/roll_int_gain: 2.0
 * /firefly/PID_attitude_controller/rotor_force_constant: 8.54858e-06
 * /firefly/PID_attitude_controller/rotor_moment_constant: 0.016
 * /firefly/mav_nonlinear_mpc/KF_observer/P0_angular_velocity: 0.1
 * /firefly/mav_nonlinear_mpc/KF_observer/P0_attitude: 0.1
 * /firefly/mav_nonlinear_mpc/KF_observer/P0_force: 0.1
 * /firefly/mav_nonlinear_mpc/KF_observer/P0_position: 0.1
 * /firefly/mav_nonlinear_mpc/KF_observer/P0_torque: 0.1
 * /firefly/mav_nonlinear_mpc/KF_observer/P0_velocity: 0.1
 * /firefly/mav_nonlinear_mpc/KF_observer/calibration_duration: 5
 * /firefly/mav_nonlinear_mpc/KF_observer/drag_coefficients: [0.01, 0.01, 0]
 * /firefly/mav_nonlinear_mpc/KF_observer/external_forces_limit: [5.0, 5.0, 3.0]
 * /firefly/mav_nonlinear_mpc/KF_observer/external_moments_limit: [20.0, 20.0, 20.0]
 * /firefly/mav_nonlinear_mpc/KF_observer/omega_limit: [3.0, 3.0, 2.0]
 * /firefly/mav_nonlinear_mpc/KF_observer/pitch_damping: 0.76
 * /firefly/mav_nonlinear_mpc/KF_observer/pitch_gain: 0.84
 * /firefly/mav_nonlinear_mpc/KF_observer/pitch_omega: 7.07
 * /firefly/mav_nonlinear_mpc/KF_observer/q_angular_velocity: 0.02
 * /firefly/mav_nonlinear_mpc/KF_observer/q_attitude: 0.015
 * /firefly/mav_nonlinear_mpc/KF_observer/q_force: 0.1
 * /firefly/mav_nonlinear_mpc/KF_observer/q_position: 0.01
 * /firefly/mav_nonlinear_mpc/KF_observer/q_torque: 0.1
 * /firefly/mav_nonlinear_mpc/KF_observer/q_velocity: 0.025
 * /firefly/mav_nonlinear_mpc/KF_observer/r_attitude: 0.01
 * /firefly/mav_nonlinear_mpc/KF_observer/r_position: 0.001
 * /firefly/mav_nonlinear_mpc/KF_observer/r_velocity: 0.0012
 * /firefly/mav_nonlinear_mpc/KF_observer/roll_damping: 0.72
 * /firefly/mav_nonlinear_mpc/KF_observer/roll_gain: 0.8
 * /firefly/mav_nonlinear_mpc/KF_observer/roll_omega: 7.07
 * /firefly/mav_nonlinear_mpc/KF_observer/yaw_damping: 0.95
 * /firefly/mav_nonlinear_mpc/KF_observer/yaw_gain: 1.0
 * /firefly/mav_nonlinear_mpc/KF_observer/yaw_omega: 5.0
 * /firefly/mav_nonlinear_mpc/K_yaw: 1.8
 * /firefly/mav_nonlinear_mpc/Ki_xy: 0.2
 * /firefly/mav_nonlinear_mpc/Ki_z: 0.3
 * /firefly/mav_nonlinear_mpc/antiwindup_ball: 0.4
 * /firefly/mav_nonlinear_mpc/enable_integrator: True
 * /firefly/mav_nonlinear_mpc/enable_offset_free: True
 * /firefly/mav_nonlinear_mpc/linear_drag_coefficients: [0.01, 0.01, 0]
 * /firefly/mav_nonlinear_mpc/mass: 1.56
 * /firefly/mav_nonlinear_mpc/pitch_gain: 0.78
[... 292 lines cut ...]
    import pkg_resources
  File "/home/dlvirmac/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3251, in <module>
    @_call_aside
  File "/home/dlvirmac/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3235, in _call_aside
    f(*args, **kwargs)
  File "/home/dlvirmac/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3264, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/home/dlvirmac/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 574, in _build_master
    ws = cls()
  File "/home/dlvirmac/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 567, in __init__
    self.add_entry(entry)
  File "/home/dlvirmac/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 623, in add_entry
    for dist in find_distributions(entry, True):
  File "/home/dlvirmac/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2065, in find_on_path
    for dist in factory(fullpath):
  File "/home/dlvirmac/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2127, in distributions_from_metadata
    if len(os.listdir(path)) == 0:
KeyboardInterrupt
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
[image]
[image]

Schmluk, 2024-01-12:
Hi @zcc0514 

Have you installed all dependencies, built them, and source your workspace? It looks like ros can not find your nodes.

zcc0514, 2024-01-13:
Then I would like to ask, how to solve my problem? Because I'm a novice, I'm new to ros. I hope you can give me some good suggestions, and thank you sincerely for your reply again!

Schmluk, 2024-01-23:
Have you tried `source ~/catkinw_ws/devel/.bashrc`?

There's many great ROS tutorials available online (e.g. http://wiki.ros.org/ROS/Tutorials). I'd suggest familiarizing yourself with ROS and let us know if your errors persist!