ethz-asl/mav_active_3d_planning #63: Stable Ubuntu and Airsim Version

GAEUNYIM, 2024-01-18:
Hi,

I appreciate for your wonderful work!
It works in my computer really well now :)

For now, Active3D is stable in the low versions following the information below.
- Ubuntu 16.04
- Airsim 4.16
- UnrealCV 0.3.10

As far as I know, your dataset and UnrealCV are stable at most Airsim version 4.16.
Also, Airsim 4.16 is available with no small errors (cmake, or compile) at most Ubuntu 16.04 for me.

However, I want to try to setup upper version of the environment setup (ex. Ubuntu 20.04, Airsim 4.27), but having troubles. 
Which versions of Ubuntu and Airsim did you use when you setup the environment?
Could you give me some information?

Waiting for your help....!
Again, thank you for the nice work, and your kind replies in github issues.

Schmluk, 2024-01-23:
Hi @GAEUNYIM 

Thanks for your interest! 

We've used Ubuntu 16.04 + Ros Kinetic with UE 4.16.3 to create the data. We've updated some of the data with newer simulators, I strongly suggest you give this a try instead as it uses AirSim as interface and has more features and better compatibility: https://github.com/ethz-asl/unreal_airsim

We've tested these datasets with Ubuntu 18.04 + Ros Melodic and UE 4.25. While most features should be portable, UE is unfortunately not well known for version compatibility, so we can not offer additional support here. Much success with setting up new environments!

GAEUNYIM, 2024-01-24:
Hello @Schmluk 

Thank you for your kind reply and suggestion of a new detailed instruction with `unreal_airsim` repo!

I will give it a try to re-install Ubuntu 18.03 + Ros Melodic and UE 4.25 and `unreal_airsim` with your datasets. Since the entire environment setup was not easy for me, it will be another new long journey :) Let me catch up step by step!

wangjinhoon, 2024-02-14:
Hello @Schmluk

Thank you for providing the code.
I tried to run the code you provided with UE 4.25 on Ubuntu 20.04. I confirmed that "roslaunch unreal_airsim demo.launch" works fine. 

However, when I try "roslaunch active_3d_planning_app_reconstruction example.launch", I get the following error:

"No unreal game running to connect to. Please start a game before launching the node.
ERROR:init:153:Fail to send message, client is not connected."

Is this issue due to unreal_cv_ros not being compatible with UE 4.25?

Is there a way to run "roslaunch active_3d_planning_app_reconstruction example.launch" without using unreal_cv_ros?

thank your reply!!

Schmluk, 2024-02-14:
> Isn't unreal_airsim replacing unreal_cv_ros?

Yes, you need to also replace the unreal_cv_ros node with an unreal_airsim node and remap the topics of the planner accordingly. You can find an example of this in [GLocal](https://github.com/ethz-asl/glocal_exploration/blob/master/glocal_exploration_ros/launch/run_glocal.launch).

wangjinhoon, 2024-02-26:
Hello @Schmluk

I tried modifying it as you suggested. I replaced "unreal_cv_ros node" with "unreal_airsim node" and remapped the planner's topic following the example of "run_glocal."

After that, when I run "roslaunch active_3d_planning_app_reconstruction airsim_example.launch planner_config:=planners/reconstruction_planner.yaml," I encounter the following issue and it terminates. Could you help me with this?

[image]

Below is the launch file I modified.
https://drive.google.com/file/d/1ocCyzYjkVVkQrjVRBiydYD4CL4gwMgrn/view?usp=sharing

And the City Building data doesn't have City Building_AirSim, so is it correct that I only need to put Blocks' Plugins folder in the project folder?

thank you

Schmluk, 2024-03-20:
Hi @wangjinhoon 

Apologies for the late reply. Yes, simply porting the plugin should work for any UE4 project! (as long as the compiled plugin's version is compatible)

W.r.t. no odometry being received, I'm not sure the simulator publishes odometry as a message or tf only, IIRC there is a param to enable this, and `rostopic list` should tell you which topics Active3D needs to subscribe to. If odometry via message is not supported (which should not be the case), adding an odometry-msg publisher next to the tf should be easy. Hope this helps!

wangjinhoon, 2024-03-22:
Hello @Schmluk
Thank you for taking the time to respond despite being busy. I will try again using the method you suggested :)

sherrypixel, 2024-11-14:
@wangjinhoon Were you able to make it work with airsim finally?

sherrypixel, 2024-11-14:
Would it possible to share an example of mav_active_3d_planning getting implemented in AirSim or Gazebo @Schmluk . I happen to be a beginner and am seriously struggling with this over a month.

Schmluk, 2024-11-14:
Thanks for this suggestion! We will try to get an example like this setup but unfortunately, we're currently a bit spread thin, so I cannot guarantee that it will be out soon. Thank you for your understanding.