ethz-asl/mav_active_3d_planning #64: Error in utils.h when compiling mavros using catkin build

LyMing9, 2024-02-19:
Hello! Thank you very much for your project! I encountered an issue while following your installation process on GitHub. When using catkin build to compile, I received an error related to mavros. It seems that there is a redeclaration problem in the utils.h file. Could you please help me understand the cause of this error and provide a solution? Thank you very much!

### Environment Information:

Operating System: Using VMware virtual machine, Ubuntu 18.04
ROS Version: Melodic
Unreal Engine: Not installed yet

### Error Message:
In file included from /home/l/catkin_ws/src/mavros/mavros/src/lib/enum_sensor_orientation.cpp:17:0:
/home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:70:40: error: ‘std::__cxx11::string mavros::utils::to_string’ redeclared as different kind of symbol
 std::string to_string(mavlink::common::MAV_AUTOPILOT e);
                                        ^~~~~~~~~~~~~
/home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:68:13: note: previous declaration ‘std::__cxx11::string mavros::utils::to_string(mavlink::common::MAV_SENSOR_ORIENTATION)’
 std::string to_string(mavlink::common::MAV_SENSOR_ORIENTATION e);
             ^~~~~~~~~
/home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:70:40: error: ‘MAV_AUTOPILOT’ is not a member of ‘mavlink::common’
 std::string to_string(mavlink::common::MAV_AUTOPILOT e);
                                        ^~~~~~~~~~~~~
/home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:70:40: note: suggested alternative:
In file included from /opt/ros/melodic/include/mavlink/v2.0/common/../standard/standard.hpp:42:0,
                 from /opt/ros/melodic/include/mavlink/v2.0/common/common.hpp:2674,
                 from /home/l/catkin_ws/src/mavros/libmavconn/include/mavconn/mavlink_dialect.h:26,
                 from /home/l/catkin_ws/src/mavros/mavros_msgs/include/mavros_msgs/mavlink_convert.h:18,
                 from /home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:22,
                 from /home/l/catkin_ws/src/mavros/mavros/src/lib/enum_sensor_orientation.cpp:17:
/opt/ros/melodic/include/mavlink/v2.0/common/../standard/../minimal/minimal.hpp:34:12: note:   ‘mavlink::minimal::MAV_AUTOPILOT’
 enum class MAV_AUTOPILOT : uint8_t
            ^~~~~~~~~~~~~
In file included from /home/l/catkin_ws/src/mavros/mavros/src/lib/enum_sensor_orientation.cpp:17:0:
/home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:71:40: error: ‘std::__cxx11::string mavros::utils::to_string’ redeclared as different kind of symbol
 std::string to_string(mavlink::common::MAV_TYPE e);
                                        ^~~~~~~~
/home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:68:13: note: previous declaration ‘std::__cxx11::string mavros::utils::to_string(mavlink::common::MAV_SENSOR_ORIENTATION)’
 std::string to_string(mavlink::common::MAV_SENSOR_ORIENTATION e);
             ^~~~~~~~~
/home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:71:40: error: ‘MAV_TYPE’ is not a member of ‘mavlink::common’
 std::string to_string(mavlink::common::MAV_TYPE e);
                                        ^~~~~~~~
/home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:71:40: note: suggested alternative: ‘AIS_TYPE’
 std::string to_string(mavlink::common::MAV_TYPE e);
                                        ^~~~~~~~
                                        AIS_TYPE
/home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:72:40: error: ‘std::__cxx11::string mavros::utils::to_string’ redeclared as different kind of symbol
 std::string to_string(mavlink::common::MAV_STATE e);
                                        ^~~~~~~~~
/home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:68:13: note: previous declaration ‘std::__cxx11::string mavros::utils::to_string(mavlink::common::MAV_SENSOR_ORIENTATION)’
 std::string to_string(mavlink::common::MAV_SENSOR_ORIENTATION e);
             ^~~~~~~~~
/home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:72:40: error: ‘MAV_STATE’ is not a member of ‘mavlink::common’
 std::string to_string(mavlink::common::MAV_STATE e);
                                        ^~~~~~~~~
/home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:72:40: note: suggested alternative: ‘MAV_FRAME’
 std::string to_string(mavlink::common::MAV_STATE e);
                                        ^~~~~~~~~
                                        MAV_FRAME
/home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:73:40: error: ‘std::__cxx11::string mavros::utils::to_string’ redeclared as different kind of symbol
 std::string to_string(mavlink::common::MAV_COMPONENT e);
                                        ^~~~~~~~~~~~~
/home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:68:13: note: previous declaration ‘std::__cxx11::string mavros::utils::to_string(mavlink::common::MAV_SENSOR_ORIENTATION)’
 std::string to_string(mavlink::common::MAV_SENSOR_ORIENTATION e);
             ^~~~~~~~~
/home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:73:40: error: ‘MAV_COMPONENT’ is not a member of ‘mavlink::common’
 std::string to_string(mavlink::common::MAV_COMPONENT e);
                                        ^~~~~~~~~~~~~
/home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:73:40: note: suggested alternative: ‘MAV_MODE’
 std::string to_string(mavlink::common::MAV_COMPONENT e);
                                        ^~~~~~~~~~~~~
                                        MAV_MODE
/home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:114:18: error: ‘MAV_TYPE’ in namespace ‘mavlink::common’ does not name a type
[... 160 lines cut ...]
 std::string to_string(GPS_FIX_TYPE e)
             ^~~~~~~~~
/home/l/catkin_ws/src/mavros/mavros/src/lib/enum_to_string.cpp:584:23: error: ‘MAV_COMPONENT’ was not declared in this scope
 std::string to_string(MAV_COMPONENT e)
                       ^~~~~~~~~~~~~
/home/l/catkin_ws/src/mavros/mavros/src/lib/enum_to_string.cpp:584:23: note: suggested alternative:
In file included from /opt/ros/melodic/include/mavlink/v2.0/common/../standard/standard.hpp:42:0,
                 from /opt/ros/melodic/include/mavlink/v2.0/common/common.hpp:2674,
                 from /home/l/catkin_ws/src/mavros/libmavconn/include/mavconn/mavlink_dialect.h:26,
                 from /home/l/catkin_ws/src/mavros/mavros_msgs/include/mavros_msgs/mavlink_convert.h:18,
                 from /home/l/catkin_ws/src/mavros/mavros/include/mavros/utils.h:22,
                 from /home/l/catkin_ws/src/mavros/mavros/src/lib/enum_to_string.cpp:19:
/opt/ros/melodic/include/mavlink/v2.0/common/../standard/../minimal/minimal.hpp:165:12: note:   ‘mavlink::minimal::MAV_COMPONENT’
 enum class MAV_COMPONENT
            ^~~~~~~~~~~~~
/home/l/catkin_ws/src/mavros/mavros/src/lib/enum_to_string.cpp:608:1: error: ‘MAV_TYPE’ does not name a type; did you mean ‘LC_CTYPE’?
 MAV_TYPE mav_type_from_str(const std::string &mav_type)
 ^~~~~~~~
 LC_CTYPE
CMakeFiles/mavros.dir/build.make:86: recipe for target 'CMakeFiles/mavros.dir/src/lib/enum_to_string.cpp.o' failed
make[2]: *** [CMakeFiles/mavros.dir/src/lib/enum_to_string.cpp.o] Error 1
CMakeFiles/Makefile2:552: recipe for target 'CMakeFiles/mavros.dir/all' failed
make[1]: *** [CMakeFiles/mavros.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

LyMing9, 2024-02-23:
Hi @wangjinhoon ,

Thank you so much for resolving my issue on GitHub! It indeed worked for me! I updated the mavros version from 0.33.3 to the latest version 1.17.0, and the project was successfully built. Thanks again for your time!

Schmluk, 2024-12-11:
Great that this worked! As an additional side note, AFAIK mavros is only used in active_3d_planning_mav to generate smooth output trajectories. That package and mavros can safely be disabled / catkin_ignored in most cases.

closed 2024-12-11 by Schmluk